TaylorHokanson/CAAint #1: Cannot program while attached to PCB

TaylorHokanson, 2018-05-13:
Seems like this is a known issue: https://github.com/espressif/arduino-esp32/issues/409

TaylorHokanson, 2018-05-15:
This looks promising:

"The same goes with GPIO pins 0, 1, 2, and 3. As far as I can tell, these pins should not be used for general purpose GPIO. **If you use 0 or 2 I understand you risk screwing up the boot of the board.** (I suppose you could use them for output after the board is booted. And I suppose you could use them for input, but you better not have anything connected before the processor boots.) If you use 1 or 3, you screw up the UART and you cannot monitor the board. (I suppose there is a use case where you don't need to monitor the board....probably after the board goes into production....if you trust never being able to monitor a board once the design has gone into production and the UART pins are used as GPIO.)"

pawelcern, 2018-07-24:
Hi. I had similar problem with custom board based on ESP-WROOM-32 even when all SPI related GPIOs were free. The problem was I had GPIO12 pulled up at boot. According to ESP32 datasheet, GPIO12 (MTDI) is somehow responsible for FLASH voltage.